heitzmann/gdstk #233: Pin Routing

Rishabhgoyal07, 2024-01-25:
Hello @heitzmann, Is pin routing possible in this module?? As i explained about pins earlier, pins are brought from leaf cell level to the top level, when this top level instance go to PnR Tool, they have pin pitches, so we have to match the pin at top level with these pin pitches or adjust them according to the pin pitch(multiples of pin pitches)

Eg; pin pitch = 0.18
Pin at the leaf cell level = 0.34
Then, we have to place this pin at the top level in multiples of pin pitches here it will be 0.36, So we have to route this pin to 0.36 at the top level.

Is it possible to do?

heitzmann, 2024-01-26:
I'm not sure I understand exactly what you need to do, but you should be able to move references freely to match the pitch.
If you can give a minimal concrete example (sample geometry and code), maybe I can help further.

Rishabhgoyal07, 2024-02-10:
Hii @heitzmann, Is there any way in gdstk to get the **absolute coordinates of a cell**. I have created 3 hierarchy level, i..e i used 1 cell and referenced it n times to create a new cell, then used this new cell and referenced it to create another new cell, now when i want to get the coordinates of the initial cell, it gives me the normal coordinates, but i want the absolute coordinates(coordinates of the cell w.r.t to the last cell. How can i get the absolute coordinates?

Rishabhgoyal07, 2024-02-27:
Does gdstk support any auto routing techniques to connect two metals, or pins?

heitzmann, 2024-02-29:
No, there's no auto-routing in gdstk.

closed 2024-03-11 by Rishabhgoyal07