heitzmann/gdstk #234: gdstk.Library.cells does not show cell names

GaN-T, 2024-01-26:
Hello, 

I want to know if there is a way to retrieve the cell names from the loaded GDS library?

This was straightforward using gdspy, as calling: 

`gdspy.library.GdsLibrary.cells`

 would return a dictionary with the cell name as the key and the Cell object as a value.

However with gdstk, calling:

`gdstk.Library.cells`

 only returns a list with the cell objects without the associated cell names. 

I am trying to use only certain cells in my gds file for a different purpose e.g. reuse in a different project. It isn't clear to me how I can do that as I do not know which cell object corresponds to which cell name. I would like to take advantage of the superior performance of gdstk over gdspy, but this is a major obstacle for me. Any suggestions would be helpful, perhaps I have missed the obvious, but I haven't  yet come across the method to access cells by their names in gdstk.

Thanks

heitzmann, 2024-01-26:
We don't return the cells in a dictionary by names because they can easily become invalid, if you change the cell name, but it's easy to build one:
```python
cells = {c.name: c for c in my_library.cells}
```
then you can use them by name as you wish.

GaN-T, 2024-01-27:
Thank you- This works !

closed 2024-01-27 by GaN-T