heitzmann/gdstk #237: How to create nested cells using gdstk

GaN-T, 2024-02-16:
How do I create a heirarchial cell structure in gdstk? The example of cell references in the documentation results in independent cells appearing in the viewer cell tree. 

```
import gdstk
import numpy as np

lib = gdstk.Library()
# Layer/datatype definitions for each step in the fabrication
ld = {
    "full etch": {"layer": 1},
    "partial etch": {"layer": 2},
    "lift-off": {"layer": 0},
}

p1 = gdstk.rectangle((-3, -3), (3, 3), **ld["full etch"])
p2 = gdstk.rectangle((-5, -3), (-3, 3), **ld["partial etch"])
p3 = gdstk.rectangle((5, -3), (3, 3), **ld["partial etch"])
p4 = gdstk.regular_polygon((0, 0), 2, 6, **ld["lift-off"])

# Create a cell with a component that is used repeatedly
contact = lib.new_cell("CONTACT")
contact.add(p1, p2, p3, p4)

cutout = gdstk.Polygon(
    [(0, 0), (5, 0), (5, 5), (0, 5), (0, 0), (2, 2), (2, 3), (3, 3), (3, 2), (2, 2)]
)
# Create a cell with the complete device
device = lib.new_cell("DEVICE")
device.add(cutout)
# Add 2 references to the component changing size and orientation
ref1 = gdstk.Reference(contact, (3.5, 1), magnification=0.25)
ref2 = gdstk.Reference(contact, (1, 3.5), magnification=0.25, rotation=np.pi / 2)
device.add(ref1, ref2)

# The final layout has several repetitions of the complete device
main = lib.new_cell("MAIN")
main.add(gdstk.Reference(device, (0, 0), columns=3, rows=2, spacing=(6, 7)))

# lib.add(main)
lib.write_gds("test.gds")
```

How do we achieve a nested structure e.g. shown [here](https://www.klayout.de/forum/discussion/2155/how-to-create-a-new-cell-in-a-layout-nested-within-another-cell)

Rishabhgoyal07, 2024-02-17:
@GaN-T, you have to reference the cells to create nested cell.

GaN-T, 2024-02-17:
Could you please show hoe the example code I show would be modified to achieve this?

Patarimi, 2024-02-19:
Hello everyone,
I think I run into the same problem. I try the following code :
```python
lib = gdstk.Library("transfo")
cell1 = octagonal_inductor() # return a gdstk.Cell
transfo = lib.new_cell("transfo")
transfo.add(gdstk.Reference(cell1))
transfo.write_svg("ind.svg") # This contains the drawing
lib.write_gds("ind.gds") # This only contains empty cells
```
I also try adding `lib.add(cell1)` before and after adding the reference (line 4). Both give a gds with empty cells and a correct svg file.
Is this a bug or is there an error in my code ?

Rishabhgoyal07, 2024-02-21:
in the **lib.add(main,*main.dependencies(True)**), this will add all the dependencies of the referenced cell, and all the cells will be shown with their hierarchy

GaN-T, 2024-02-21:
this didn't change the output from the example I initially provided referencing the "Getting started" documentation. The heirarchy is there, it is just not shown in the KLayout Cell viewer. Please see [this link](https://www.klayout.de/forum/discussion/2155/how-to-create-a-new-cell-in-a-layout-nested-within-another-cell) for an example of what I am trying to do

Rishabhgoyal07, 2024-02-21:
paste your klayout screenshot

GaN-T, 2024-02-21:
It doesn't appear as a nested tree like I want even though the hierarchy is present in the layout. When I flatten cell list I can see the components in the list too, but as individual cells, not nested below main. 
```

import gdstk
import numpy as np

lib = gdstk.Library()
# Layer/datatype definitions for each step in the fabrication
ld = {
    "full etch": {"layer": 1},
    "partial etch": {"layer": 2},
    "lift-off": {"layer": 0},
}

p1 = gdstk.rectangle((-3, -3), (3, 3), **ld["full etch"])
p2 = gdstk.rectangle((-5, -3), (-3, 3), **ld["partial etch"])
p3 = gdstk.rectangle((5, -3), (3, 3), **ld["partial etch"])
p4 = gdstk.regular_polygon((0, 0), 2, 6, **ld["lift-off"])

# Create a cell with a component that is used repeatedly
contact = lib.new_cell("CONTACT")
contact.add(p1, p2, p3, p4)

cutout = gdstk.Polygon(
    [(0, 0), (5, 0), (5, 5), (0, 5), (0, 0), (2, 2), (2, 3), (3, 3), (3, 2), (2, 2)]
)
# Create a cell with the complete device
device = lib.new_cell("DEVICE")
device.add(cutout)
# Add 2 references to the component changing size and orientation
ref1 = gdstk.Reference(contact, (3.5, 1), magnification=0.25)
ref2 = gdstk.Reference(contact, (1, 3.5), magnification=0.25, rotation=np.pi / 2)
device.add(ref1, ref2)

# The final layout has several repetitions of the complete device
main = lib.new_cell("MAIN")
main.add(gdstk.Reference(device, (0, 0), columns=3, rows=2, spacing=(6, 7)))

lib.add(main, *main.dependencies(True))
lib.write_gds("test.gds")
```

Could you post your output?

nmz787-intel, 2024-03-07:
Using the original/first post code, I get this output
 [image]

You may need to right-click the cell list, and then uncheck "Flat Cell List"
[image]

You may also notice that you can select which cell you want to start viewing the hierarchy from, with the so-called "Show As New Top"

GaN-T, 2024-03-07:
Thanks this is exactly it!

closed 2024-03-07 by GaN-T