heitzmann/gdstk #238: Missing Python 3.12 wheel file for Windows

github-young, 2024-02-21:
There is no wheel file for Python 312 on the Windows platform on [PyPI](https://pypi.org/project/gdstk/#files) by the time this issue was filed.

It turns out that the `"3.12"` term is missing in the [GitHub Actions file](https://github.com/heitzmann/gdstk/blob/b7614f2a7bf26b52c2bee787451af5539ffeb7be/.github/workflows/pypi-packages.yml#L46).

```yml
python-version: ["3.8", "3.9", "3.10", "3.11"]
```

Could you please add this missing file? Thank you.

heitzmann, 2024-02-21:
Python 3.12 is not working on Windows in Github actions: https://github.com/heitzmann/gdstk/actions/runs/7990757750/job/21820267931

If you know any solution to this issue, feel free to open a PR!

closed 2024-02-21 by heitzmann

github-young, 2024-02-22:
In addition to the [pip-tools](https://github.com/jazzband/pip-tools/issues/2021) issue, [this issue](https://github.com/conan-io/conan/issues/14801) says conan 1.x is not compatible with Python 3.12 either.