heitzmann/gdstk #244: Python tests fail

yurivict, 2024-04-18:
```
cd /usr/ports/cad/py-gdstk/work-py39/gdstk-0.9.51 && /usr/bin/env -i HOME=/usr/ports/cad/py-gdstk/work-py39  PWD="${PWD}"  __MAKE_CONF=/nonexistent OSVERSION=1400509 PATH=/usr/local/libexec/ccache:/usr/ports/cad/py-gdstk/work-py39/.bin:/home/yuri/bin:/sbin:/bin:/usr/sbin:/usr/bin:/usr/local/sbin:/usr/local/bin TERM=xterm-256color XDG_DATA_HOME=/usr/ports/cad/py-gdstk/work-py39  XDG_CONFIG_HOME=/usr/ports/cad/py-gdstk/work-py39  XDG_CACHE_HOME=/usr/ports/cad/py-gdstk/work-py39/.cache  HOME=/usr/ports/cad/py-gdstk/work-py39 PATH=/usr/local/libexec/ccache:/usr/ports/cad/py-gdstk/work-py39/.bin:/home/yuri/bin:/sbin:/bin:/usr/sbin:/usr/bin:/usr/local/sbin:/usr/local/bin PKG_CONFIG_LIBDIR=/usr/ports/cad/py-gdstk/work-py39/.pkgconfig:/usr/local/libdata/pkgconfig:/usr/local/share/pkgconfig:/usr/libdata/pkgconfig MK_DEBUG_FILES=no MK_KERNEL_SYMBOLS=no SHELL=/bin/sh NO_LINT=YES PREFIX=/usr/local  LOCALBASE=/usr/local  CC="cc" CFLAGS="-O2 -pipe  -fstack-protector-strong -fno-strict-aliasing "  CPP="cpp" CPPFLAGS=""  LDFLAGS=" -fstack-protector-strong " LIBS=""  CXX="c++" CXXFLAGS="-O2 -pipe -fstack-protector-strong -fno-strict-aliasing  " CCACHE_DIR="/tmp/.ccache" BSD_INSTALL_PROGRAM="install  -s -m 555"  BSD_INSTALL_LIB="install  -s -m 0644"  BSD_INSTALL_SCRIPT="install  -m 555"  BSD_INSTALL_DATA="install  -m 0644"  BSD_INSTALL_MAN="install  -m 444" PYTHONPATH=/usr/ports/cad/py-gdstk/work-py39/stage/usr/local/lib/python3.9/site-packages /usr/local/bin/python3.9 -m pytest -k '' -rs -v -o addopts= 
/usr/local/lib/python3.9/site-packages/_pytest/config/__init__.py:331: PluggyTeardownRaisedWarning: A plugin raised an exception during an old-style hookwrapper teardown.
Plugin: helpconfig, Hook: pytest_cmdline_parse
ConftestImportFailure: ModuleNotFoundError: No module named 'gdstk._gdstk' (from /usr/ports/cad/py-gdstk/work-py39/gdstk-0.9.51/tests/conftest.py)
For more information see https://pluggy.readthedocs.io/en/stable/api_reference.html#pluggy.PluggyTeardownRaisedWarning
  config = pluginmanager.hook.pytest_cmdline_parse(
ImportError while loading conftest '/usr/ports/cad/py-gdstk/work-py39/gdstk-0.9.51/tests/conftest.py'.
tests/conftest.py:12: in <module>
    import gdstk
gdstk/__init__.py:1: in <module>
    from ._gdstk import *
E   ModuleNotFoundError: No module named 'gdstk._gdstk'
*** Error code 4
```

Version: 0.9.51
Python-3.9
FreeBSD 13.2

heitzmann, 2024-04-18:
Is there a _gdstk-*.so in the gdstk directory?
Did you compile it?

yurivict, 2024-04-18:
This file is present:
/usr/local/lib/python3.9/site-packages/gdstk/_gdstk.cpython-39.so

heitzmann, 2024-04-18:
In that case, are you trying to run the tests from the source distribution directory? If so, your python interpreter will look for the local directory _gdstk_ instead of _/usr/local/lib/python3.9/site-packages/gdstk_, which probably does not contain the so.

For local development I usually create a symlink from the build directory to _gdtsk_. If you have a better alternative, let me know.

closed 2024-07-04 by heitzmann