heitzmann/gdstk #245: 0.9.51: Doesn't install include directory and .pc file

yurivict, 2024-04-18:
These lines disappeared:
```
install(FILES ${CMAKE_CURRENT_BINARY_DIR}/gdstk.pc DESTINATION ${CMAKE_INSTALL_LIBDIR}/pkgconfig)
CONFIGURE_FILE("gdstk.pc.in" "gdstk.pc")
```

This line also disappeared:
```
include(GNUInstallDirs)
```
It probably installed the ```include``` directory.

heitzmann, 2024-04-18:
That's right, the whole build system changed because there were too many hacks. The gdstk.pc was one of them. It was not properly created and was starting to break on Windows.

The headers I forgot to check after the changes, but I've put them back in the latest commit.

yurivict, 2024-04-28:
Do you intend to make a corrected release?

heitzmann, 2024-05-24:
Sorry, I thought I had already released this... I've prepared the release locally and completely forgot to push!

closed 2024-07-04 by heitzmann