heitzmann/gdstk #248: gdstk Boolean too slow but works fast in Klayout??

GaN-T, 2024-05-02:
I am trying to cut out several holes in a rectangle. When I do this in Klayout using "Subtract others from first", the result is computed almost instantly. However, when I try to do it using gdstk, it is very slow! Am i doing something wrong?

GDSTK code:

 'box' is the rectangle i want to cut the holes in
 'pols' is a list of 2713 polygons i want to cut out of 'box'

```
res = box
for i in range(len(pols)):
    if res!= []:
        res = gdstk.boolean(res, pols[i], 'not')[0]
```

The Klayout GUI boolean operation works very fast somehow as I am selecting all the polygons to be cut at once with multiple selection. I suspect this is different from the sequential for loop update for each cut operation im doing using gdstk. Is there a smarter + faster way to do this in gdstk? Many thanks for your help!

heitzmann, 2024-05-02:
Yes: `res = gdstk.boolean(box, pols, 'not')`

GaN-T, 2024-05-02:
this is what i tried initially, but the result is an empty list

heitzmann, 2024-05-02:
Then there is probably something wrong with your inputs. If you can share the example I can take a look.

GaN-T, 2024-05-02:
```

def getCellExtents2(cell , precision:int = 6):
    
    if isinstance(cell,gdstk.Cell) or isinstance(cell,gdstk.Polygon):
        cellBb = np.around(cell.bounding_box(), precision) 
    else:
        cellBb = np.around(cell.bounding_box(), precision) 
    (x1,y1) = (cellBb[0][0], cellBb[0][1])
    (x2,y2) = (cellBb[1][0], cellBb[1][1])
    cellW = abs(cellBb[0][0] - cellBb[1][0])   
    cellL = abs(cellBb[0][1] - cellBb[1][1])  
    if x1 > x2:
        xc = (x1 - x2)/2
    else:
        xc = -(x1 - x2)/2
    if y1 > y2:
        yc = (y1 - y2)/2
    else:
        yc = -(y1 - y2)/2
    return cellBb, (xc, yc), cellW, cellL

class RegularPolyPost():

    def __init__(self, lib, sideLen, nSides, layer = 0, rotation = 0):
        self.lib = lib
        self.d = sideLen               # for circles this is the diameter
        self.nSides = nSides           # number of sides of polygon
        self.layer = layer
        self.rotation = rotation       # angle in degrees
        self.name = f"{self.d}um_{self.nSides}"
        self.cell = lib.new_cell(f"{self.name}")

        if self.nSides == 0:
            post = gdstk.ellipse((0,0), self.d/2, tolerance=0.01, layer = self.layer)
        else:
            post = gdstk.regular_polygon((0,0), self.d, self.nSides, np.radians(self.rotation), layer = self.layer)
        self.cell.add(post)

class DLDFlowCell():

    def __init__(self, lib, dc, eps, ioDiam, ioL, ioW, ioAngle, portAngle, fillet, cW, cL, pD , name = "default"):

        self.lib = lib
        self.dc = dc                              # particle diameter
        self.eps = eps                            # row shift parameter
        self.ioDiam = ioDiam                      # io port diameter
        self.ioL = ioL                            # io channel length 
        self.ioW = ioW                            # io channel width
        self.ioAngle = ioAngle                    # io channel angle 
        self.cW = cW                              # dld channel width  
        self.cL = cL                              # dld channel length  
        self.pD = pD                              # post object -> must be a gdstk cell or polygon
        self.g = self.dc/(1.4*(self.eps)**0.48)   # nominal spacing between posts - McGrath, Lab Chip, 2014, 14, 4139–4158.                             
        self.name = name
        self.fillet = fillet                      # fillet on the IO channel branch off point
        self.portAngle = portAngle                # angle subtended by overlapping chord of io ports on channels
        self.createFlowCell()
        

    def createFlowCell(self):

        if self.name == "default":
            self.name = f"DLD_{self.cL}_{self.cW}_{self.cL}_"
        self.cell = lib.new_cell(f"{self.name}")
        self.dldChannel = gdstk.rectangle((-self.cL/2,-self.cW/2),(self.cL/2,self.cW/2))

        self.dldRegion = self.dldChannel.copy()
        self.fillDldPattern(correction = False)
        self.invertDld()
        # self.dldChannel = gdstk.boolean(self.dldChannel, gdstk.Reference(self.clip), 'not')[0] <<<<<<<<<<<<
        ioL = gdstk.Polygon([(-self.ioL/2,-self.ioW/2),(self.ioL/2,-self.cW/2),
                             (self.ioL/2,self.cW/2),(-self.ioL/2,self.ioW/2)])\
                             .translate(-self.ioL/2-self.cL/2,0)
        port1 = gdstk.ellipse((-self.ioL - self.cL/2 - self.ioDiam/2*np.sin(np.radians(self.portAngle)),0),
                               self.ioDiam/2)
        ioL = gdstk.boolean(ioL, port1, 'or')[0]

        io1 = gdstk.rectangle((-self.ioL/2,-self.ioW/2),(self.ioL/2,self.ioW/2))
        port2= gdstk.ellipse((self.ioL/2 + self.ioDiam/2*np.sin(np.radians(self.portAngle)),0),
                              self.ioDiam/2)
        yOff = -io1.points[0][1]
        xOff = -io1.points[0][0] + self.cL/2
        union = gdstk.boolean(io1, port2, "or")[0]
        io1 = union.translate(xOff,yOff).rotate(np.radians(self.ioAngle/2),
                                                 center = (self.cL/2, 0))
        io2 = io1.copy()
        io2 = io2.mirror((0,0),(1,0))
        ioR = gdstk.boolean(io1, [io2, self.dldChannel], 'or')[0]
        
        self.dldChannel = gdstk.boolean(ioL, ioR, 'or')[0].fillet(self.fillet)
        self.cell.add(self.dldChannel)
    

    def fillDldPattern(self, correction = False):
        
        pRow = self.lib.new_cell("pArray")
        self.pRef = self.lib.new_cell("pRef")
        self.clip = self.lib.new_cell("clip")
        _,_,pL,pW = getCellExtents2(self.pD.cell)
        ny = int(self.cL / (self.g + pW)) + 1  # ncols
        nx = int(self.cW / (self.g + pW)) + 1 # nrows
        spc = pW + self.g
        extras = int(ny*self.eps) + 1
        for i in range(ny):
            arr = gdstk.Reference(self.pD.cell, origin = (i*spc,i*self.eps*spc - extras*spc), columns = 1, rows= nx + extras, spacing = (0,spc))
            pRow.add(arr)     
        rowRef = gdstk.Reference(pRow, origin = (-self.cL/2,-self.cW/2), columns = 1, rows= 1, spacing = (0,0))
        self.pRef.add(rowRef)
        self.mask = gdstk.boolean(gdstk.offset(self.dldRegion, (extras+2)*spc)[0], self.dldRegion, 'not')[0]
        self.pols = self.pRef.flatten().polygons
        for i in range(len(self.pols)):
            res = gdstk.boolean(self.pols[i], self.mask, 'not')
            if res != []:
                if res[0].area() > 25:
                    self.clip.add(res[0])
        self.clip.add(self.dldRegion)

        lib.remove(self.pRef, pRow, self.pD.cell)
        
            
    def invertDld(self):
        clipFlat = self.clip.flatten()
        self.pols = clipFlat.polygons
        self.inv = self.lib.new_cell("invDld")
        self.holes = []
        self.res = self.dldRegion

        self.res = gdstk.boolean(self.res,self.pols,'not')
            
                
       

```

GaN-T, 2024-05-02:
its the last function invertDld. I had the same problem in creating the clip cell too. I only got the output i needed when i did it in the for loop within fillDldPattern():
```

for i in range(len(self.pols)):
            res = gdstk.boolean(self.pols[i], self.mask, 'not')
            if res != []:
                if res[0].area() > 25:
                    self.clip.add(res[0])
        self.clip.add(self.dldRegion)
```

GaN-T, 2024-05-02:
i can try to make a more minimal working example, give me a moment

GaN-T, 2024-05-02:
```
import gdstk

dc =5.8
eps = 1/20
cW=2000
d = 70
cL = 10000
g = dc/(1.4*(eps)**0.48) 

lib = gdstk.Library()

pD = lib.new_cell("post")
pRow = lib.new_cell("pArray")
pRef = lib.new_cell("pRef")
clip = lib.new_cell("clip")
inv = lib.new_cell("invDld")

pD.add(gdstk.ellipse((0,0), d/2, tolerance=0.01))   # unit cell for array
box = gdstk.rectangle((-cL/2,-cW/2),(cL/2,cW/2))    # box to cut from

ny = int(cL / (g + d)) + 1  # ncols
nx = int(cW / (g + d)) + 1 #  nrows
spc = d + g                # array spacing
extras = int(ny*eps) + 1

# create array 
for i in range(ny):
    arr = gdstk.Reference(pD, origin = (i*spc,i*eps*spc - extras*spc), columns = 1, rows= nx + extras, spacing = (0,spc))
    pRow.add(arr) 
rowRef = gdstk.Reference(pRow, origin = (-cL/2,-cW/2), columns = 1, rows= 1, spacing = (0,0))
pRef.add(rowRef)

 # mask to trim off edges
mask = gdstk.boolean(gdstk.offset(box, (extras+2)*spc)[0], box, 'not')[0]

# remove posts that fall outside box
pols = pRef.flatten().polygons
for i in range(len(pols)):
    res = gdstk.boolean(pols[i], mask, 'not')
    if res != []:
        if res[0].area() > 25:
            clip.add(res[0])
clip.add(box)           # cell containing mask and objects to cut

lib.remove(pRef, pRow, pD)   # remove unwanted cells

clipFlat = clip.flatten()
pols = clipFlat.polygons

# perform inversion!
res = gdstk.boolean(box,pols, "not")[0]
inv.add(res)

lib.write_gds("minimalExample.gds")
```

GaN-T, 2024-05-06:
@heitzmann have you had the chance to try to run the minimal example? Do you get the same problem as me?

philstopford, 2024-05-14:
I'm a bit confused here. You add the box to the clip and then subtract it from the box in your boolean, don't you? That means you end up with no polygons at the end because you've cut the box from itself....

GaN-T, 2024-05-14:
@philstopford Thank you so much. That was exactly what I was doing wrong!

philstopford, 2024-05-14:
@GaN-T  Any chance you could post your modified simple test case? My brain is tired and being able to just run/poke something immediately would help.

GaN-T, 2024-05-15:
@philstopford I edited and updated my previously provided simplified example to get rid of the problematic line of code. You should be able to run it. My question is rather about the underlying principle of boolean "not" with large arrays of polygons. I first need to flatten the cell in order to get the result I want. It doesn't work without this, when I use the reference/ cell containing the polygons directly. You can see this in my example where I do the boolean "not" on two occasions. In both cases, it is preceded by the flatten operation. Without this, I get an empty list, which is not what i expected...

philstopford, 2024-05-15:
I'm not exactly sure what the output should be here. Is it a plane full of holes or a trimmed set of holes? If it's the latter, I get that quite nicely. The long delay I see is in writing the GDSII file, not the boolean operation itself. OASIS writes out much more quickly.

GaN-T, 2024-05-16:
the output in the cell called "clip" is the plane of trimmed polygons. The output in the cell called "inv" is the plate with holes cut using the polygons from the "clip". I see your point. The slow step is the file writing.. My last question was just about why its necessary to flatten cells before doing the "not" in each case. But its not a big problem in the end.. I will close the issue. Thanks very much for your help!

closed 2024-05-16 by GaN-T