heitzmann/gdstk #249: If adding a reference cell, why it is necessary to add this cell instance in lib???

huangwenjunlovedy, 2024-05-22:
I found that when using a reference cell, it is necessary to add the instance of the reference cell to the lib, otherwise it will fail. May I ask why this is happening? Shouldn't it be adding a top-level cell to lib? Will the referenced cells contained in that top-level cell be included in lib together

my code: 
[test2.zip](https://github.com/heitzmann/gdstk/files/15397496/test2.zip)

（1）When lib adds instances that reference cells :
[image]

[image]

（2）When lib does not add an instance referencing the cell :
[image]
[image]

heitzmann, 2024-05-24:
That's the expected behavior. That was a design choice made in early development, because it is quite easy to simply replace your line with `lib2.add(main, *main.dependencies())` and have the automatic behavior you desire.

nmz787-intel, 2024-06-18:
I believe it has to do with some tools that have many GDS files, and can load them all as a distributed library. This is an assumption though, and may be an incorrect interpretation of behavior I see in closed-source tools.

closed 2024-07-04 by heitzmann