heitzmann/gdstk #250: DLL load failed

MatthewMckee4, 2024-05-24:
After building the wheel package using

    python -m build -w

I ran:

    pip install gdstk/dist/gdstk-0.9.51-cp311-cp311-win_amd64.whl

to install and when I try to use it, it cannot find methods and classes etc inside it and shows this error.

[image]

heitzmann, 2024-05-24:
Is there any specific reason not tu use the [pre-compiled packages from PyPI](https://pypi.org/project/gdstk/)?

Compilation in Windows requires a few dependencies and the build system properly configured. If you share your build logs maybe we can help identify the issue.

MatthewMckee4, 2024-05-24:
there is some enterprise level business logic that I would like to implement in gdstk without it being pushed to the public package, so I'm trying to make some changes and build the python module myself. I previously installed zlib and qhull with vcpkg (I'm not sure if this is the correct approach, this could be where my issue is) as I was having issues with:
[image]
This is also the same issue I get when I pip install the tar.gz file in releases
So I included these at the top of the CMakeLists file which allowed me to build the python package fine, but when I pip install either the tar.gz file or the .whl file I get that error
[image]

heitzmann, 2024-05-24:
Yes, you're not successfully compiling the module, so the DLL is not generated. I don't really use Windows so I don't know how much help I can offer, but I think you'll want to have `ZLIB_INCLUDE_DIRS` and `ZLIB_LIBRARIES` (note the plural) in the in your *evironment*, not as a CMake variable: [see here](https://github.com/heitzmann/gdstk/blob/76f226a50e145b3411e3ad95e7a292fc9e71f275/src/CMakeLists.txt#L4)

closed 2024-05-31 by MatthewMckee4