heitzmann/gdstk #258: How to simply calculate the coordinates of the graph in a certain sub-cell in the top cell?

tjl0209, 2024-06-30:
In the design of the semiconductor industry, we need to extract multiple coordinates of certain cells containing mark in the top cell (we know that these marks are included in the top cell, but they may be multi-level). This is very important for the subsequent manufacturing. Does gdstk have a way to achieve this function simply?

frankxu-omni, 2024-07-01:
Is this what you want?
top_cell.get_labels()
top_cell.get_polygons()

tjl0209, 2024-07-02:
> Is this what you want? top_cell.get_labels() top_cell.get_polygons()

> Is this what you want? top_cell.get_labels() top_cell.get_polygons()

tjl0209, 2024-07-02:
> Is this what you want? top_cell.get_labels() top_cell.get_polygons()
[image]
Thank you for your reply! Although this doesn't seem to be what I want. I used a diagram to simply elaborate my requirement.

closed 2024-07-02 by tjl0209