heitzmann/gdstk #260: Unnecessary dependency on oldest-supported-numpy

wshanks, 2024-07-07:
oldest-supported-numpy was intended to be used as a build dependency before numpy 2 (which has backwards ABI compatibility). It should not be needed as a runtime dependency (just numpy). Also, since gdstk dropped support for Python 3.8, oldest-supported-numpy should not be needed as a build dependency either.

closed 2024-07-08 by heitzmann

daquinteroflex, 2024-08-30:
@heitzmann I think this might be related to the poetry resolution issue too https://github.com/flexcompute/tidy3d/pull/1913 according to https://github.com/python-poetry/poetry/issues/8830 and hence the unclear error on the lockfile resolution in https://github.com/python-poetry/poetry/issues/9659 for an empty project with numpy and gdstk as a dependency.

Suspect a new release with this change sounds like it might solve this environment issue. What do you think?

heitzmann, 2024-08-30:
> to the poetry resolution issue too flexcompute/tidy3d#1913
> python-poetry/poetry#8830
> unclear error on the lockfile resolution in python-poetry/poetry#9659

The newest release already includes this fix. 'oldest-supported-numpy' has
already been removed, or did I forget anything else?

daquinteroflex, 2024-08-30:
Hi Lucas,

[image]

I'm just going from this commit history, seems like this PR was merged after the 0.9.53 tag?

heitzmann, 2024-08-31:
My bad, I was assuming these changes had already been released! I've
triggered a new version release right now to include them.

daquinteroflex, 2024-09-03:
Thanks a lot Lucas! Indeed the poetry lockfile resolution is now working properly with this release