heitzmann/gdstk #264: GDS Hierarchy Traversing

Rishabhgoyal07, 2024-07-15:
Hii  @heitzmann, can we###  **traverse hierarchy in gdstk** and access cell present in that hierarchy. Example, I have 10 hierarchy in my GDS, can I access the bottom and top hierarchy and move from top to bottom or vice versa and access each cell present in that hierarchy.

heitzmann, 2024-07-15:
Yes, just look at `Cell.references`. For each reference you can go to the corresponding cell with `Reference.cell`, and then traverse its references and so on. Note that you might traverse the same cell more than once, if it is referenced multiple times.

nmz787-intel, 2024-10-01:
@Rishabhgoyal07  here's some code that given a topcell and a child that you want to get, but taking into account any offsets/rotations/repetitions the referencing did. NOTE I only implemented this for polygons and labels in the resulting "transformed cell". I have some comments for how to approach for FlexPath/etc... I think the polygon and label chunks of code should be easy to adapt to any other `cell` components.

the code:
```

def find_leaf_reference_offsets_and_rotations(cell, target_cell_name):
    """
    Recursively find the offsets and rotations applied to reach the target cell.

    Parameters:
    - cell: The current cell being traversed.
    - target_cell_name: The name of the target cell to find.
    - current_offset: The cumulative offset applied so far.
    - current_rotation: The cumulative rotation applied so far.

    Returns:
    - A list of lists of tuples containing the offset and rotation at each level of the hierarchy.
    """
    transformations = []
    current_cell_name = cell.name
    for ref in cell.references:
        if ref.cell.name == target_cell_name:
            # Found the target cell, add the current offset and rotation
            transformations.append((ref.origin, ref.rotation, ref.x_reflection, ref.magnification, target_cell_name))
        else:
            # Recursively search in the referenced cell
            result = find_leaf_reference_offsets_and_rotations(ref.cell, target_cell_name)
            if result:
                # If the target cell is found in the recursion, add the current level's offset and rotation
                for res in result:
                    transformations.append(((ref.origin, ref.rotation, ref.x_reflection, ref.magnification, ref.cell.name), res))
    return transformations

def rotate_point(point, angle, origin=(0, 0), round_to=3):
    """Rotate a point around a given origin by a given angle."""
    ox, oy = origin
    px, py = point

    qx = ox + math.cos(angle) * (px - ox) - math.sin(angle) * (py - oy)
    qy = oy + math.sin(angle) * (px - ox) + math.cos(angle) * (py - oy)
    return [round(qx, round_to), round(qy, round_to)]

def reflect_point(point, origin=(0, 0)):
    """Reflect a point across the x-axis around a given origin."""
    ox, oy = origin
    px, py = point

    qx = 2 * ox - px
    return [qx, py]

def apply_reps(cell_component_with_repetition, keep_reps_attr=False):
    reps_out = []
    if str(cell_component_with_repetition.repetition) != 'No repetition':
        rep_obj = cell_component_with_repetition.repetition
        reps = cell_component_with_repetition.apply_repetition()

        # ensure input is not in output, as I experienced once
        if isinstance(cell_component_with_repetition, gdstk.Polygon):
            polygon = cell_component_with_repetition
            reps_out += [rep for rep in reps if not (polygon.points == rep.points).all()]
        elif isinstance(cell_component_with_repetition, gdstk.Label):
            lbl = cell_component_with_repetition
            reps_out += [rep for rep in reps if (
                                                (lbl.origin != rep.origin) or
                                                (lbl.text != rep.text) or
                                                (lbl.texttype == rep.texttype) or
                                                (lbl.anchor == rep.anchor) or
                                                (lbl.rotation == rep.rotation)  or
                                                (lbl.magnification == rep.magnification) or
                                                (lbl.x_reflection == rep.x_reflection)
[... 127 lines cut ...]
            new_bb[0][0] += origin_transform[0]
            new_bb[1][0] += origin_transform[0]
            new_bb[0][1] += origin_transform[1]
            new_bb[1][1] += origin_transform[1]
            # Apply rotation around the origin
            ll = rotate_point(new_bb[0], rotation_transform, new_bb[0])
            ur = rotate_point(new_bb[1], rotation_transform, new_bb[0])
            # Apply rotation around the origin
            new_bb[0] = min(ll, ur)
            new_bb[1] = max(ll, ur)

        original_poly_transformed.append((tuple(new_bb[0]), tuple(new_bb[1])))
    assert original_poly_transformed == flat_poly_bbs

    # Apply transformations to the LEAF cell and create new transformed cells
    leaf_cell = [cell for cell in lib.cells if cell.name == 'LEAF'][0]
    transformed_cells = [apply_transformations_to_cell(leaf_cell, transform_list) for transform_list in result]

    # Check that the transformed cells match the flattened cell
    for transformed_cell in transformed_cells:
        transformed_polys = transformed_cell.get_polygons(datatype=99, layer=101)
        assert len(transformed_polys) == 1
        transformed_poly = transformed_polys[0]
        transformed_poly.bounding_box() in flat_poly_bbs
 ```