heitzmann/gdstk #266: OASIS reader doesn't update geometry-w/geometry-h to the bounding box of CTRAPEZOIDs

rocallahan, 2024-07-30:
I don't have the actual spec at hand but the [draft spec](http://thuime.cn/wiki/images/e/ea/Oasis-3626-042303-draft.pdf) says:

> 29.7 Each successive CTRAPEZOID record updates all ctrapezoid-related modal variables with the following
exception: for the forms where only one of width or height is used (types 16-23 and 25), modal variables geometry-w or geometry-h are **both updated** to match the specified dimension.

As far as I can tell, [the code](https://github.com/heitzmann/gdstk/blob/153c444b3386e2e5cb303565c719b9fc38f674dd/src/library.cpp#L2253) does not do this. This can result in subtle errors in the geometry of shapes following certain CTRAPEZOIDs in the file.

rocallahan, 2024-07-30:
It's not 100% clear in this draft spec whether "the specified dimension" means the value of geometry-w/geometry-h that's used to size the CTRAPEZOID, or the actual bounding box size of the CTRAPEZOID, but KLayout [interprets](https://github.com/KLayout/klayout/blob/cfa0dacd4fd47c9d27ee79eb4a6d163590a717bf/src/plugins/streamers/oasis/db_plugin/dbOASISReader.cc#L3024) it as the bounding box size.

heitzmann, 2024-08-09:
You need the qhull library installed on your system and available for CMake to find. That's the error being reported.

heitzmann, 2024-08-09:
Thanks for the report! You're right, we're missing a few modal variable updates for specific CTRAPEZOID types. I've added a fix in 64083fd. If you have an example where you see the error, would you mind double checking to make sure it works, @rocallahan ?

rocallahan, 2024-08-09:
I can't test it right now, but FWIW your code differs from KLayout's interpretation and I don't know who's right. E.g. for CTRAPEZOID types where the width of the bounding box is 2 x geometry-w, KLayout updates geometry-w to twice its incoming width, but your code doesn't.

heitzmann, 2024-08-15:
The changes I've added do the update to the modal dimensions in the cases when they were missing using the bounding box. Isn't that what KLayout also does?

rocallahan, 2024-08-15:
Yes, I think you're right, sorry about that.

closed 2024-08-19 by rocallahan