heitzmann/gdstk #267: Polygon.perimeter() not working?

GaN-T, 2024-07-30:
I tried using the perimeter method shown in the documentation, but it throws an Attribute error.

```
import gdstk

rect = gdstk.rectangle((0,0),(10,20))
rect.perimeter()

##################################
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[18], line 4
      1 import gdstk
      3 rect = gdstk.rectangle((0,0),(10,20))
----> 4 rect.perimeter()

AttributeError: 'gdstk.Polygon' object has no attribute 'perimeter'

```

heitzmann, 2024-08-09:
Do you have a recent version installed? I don't see the problem:
```
$ cat issue.py 
import gdstk

rect = gdstk.rectangle((0,0),(10,20))
print(rect.perimeter())

$ python issue.py 
60.0
```

nmz787-intel, 2024-08-09:
works for me too:
```
Python 3.9.6 (default, Aug 25 2021, 01:57:02) 
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gdstk
>>> 
>>> rect = gdstk.rectangle((0,0),(10,20))
>>> print(rect.perimeter())
60.0
>>> print(gdstk.__version__)
0.9.50

```

GaN-T, 2024-08-13:
I am using gdstk version: 0.9.49, Python 3.10.13

GaN-T, 2024-08-13:
i will close this. Not sure why the old version threw that error.

closed 2024-08-13 by GaN-T