heitzmann/gdstk #268: Missing polygons in output GDS

Rishabhgoyal07, 2024-08-02:
Hii @heitzmann, I am reading a GDS of a layout and creating a GDS of a leaf cell of that layout using command ### **lib.write_gds**, but in the output GDS, some polygons are missing from the output GDS which were present in the input GDS, why is this happening, Is GDSTK deleting some polygons?

Rishabhgoyal07, 2024-08-07:
@heitzmann I debugged the issue and found that **some small flexpath are not being read while**, i am getting all the polygons using **get_polygons(include_paths=True, layer= layer_number)** function, can you tell me why it is happening, other flex path are being read properly, but some small flex path are not being read. Is it a **### bug** in the gdstk?

heitzmann, 2024-08-09:
Did you also set the datatype appropriately in the `get_polygons` call?

I can only help if you send me a minimal reproducible example. Just by inspecting the code or looking at the current tests, I cannot find any problem.

Rishabhgoyal07, 2024-08-14:
Hii @heitzmann, This is a screenshot of the paths placement, similarly at other places the same kind of path is not being read.
If you need any more detail, kindly please tell me
[image]

nmz787-intel, 2024-08-14:
@Rishabhgoyal07  please provide a minimal script testcase. We cannot use your image as a testcase, though it is thought provoking.

Rishabhgoyal07, 2024-08-15:
Sorry, I cant share more than this, you can recreate this at your end and check, this kind of issue is coming in while reading the paths only, if you need any help in recreating then I will help, as per my knowledge this is a kind of bug, and it will be very useful, when it is resolved