heitzmann/gdstk #269: Memory leak in cell::flatten?

Roy-974, 2024-08-16:
Hello,

I'm using gdstk and tried to sanatize my code. It seems that the cell flatten function is leaking.

Here is my simple code:

```
#include <stdio.h>
#include <iostream>

#include <gdstk/gdstk.hpp>

using namespace gdstk;

Cell* grating(double period, double fill_frac, double length, double width, Tag tag,
              const char* name) {
    double x = width / 2;
    double w = period * fill_frac;
    int64_t num = (int64_t)(length / period);

    Cell* result = (Cell*)allocate_clear(sizeof(Cell));
    result->name = copy_string(name, NULL);
    result->polygon_array.ensure_slots(num);
    for (int64_t i = 0; i < num; i++) {
        double y = i * period;
        Polygon* rect = (Polygon*)allocate(sizeof(Polygon));
        *rect = rectangle(Vec2{-x, y}, Vec2{x, y + w}, tag);
        result->polygon_array.append(rect);
    }

    return result;
}

int main(int argc, char* argv[]) {
    Library lib = {};
    lib.init("library", 1e-6, 1e-9);

    double length = 20;

    Cell* grat1 = grating(3.5, 0.5, length, 25, make_tag(1, 0), "Grating 1");
    lib.cell_array.append(grat1);

    Cell* grat2 = grating(3.0, 0.5, length, 25, make_tag(1, 0), "Grating 2");
    lib.cell_array.append(grat2);

    Cell* main_cell = (Cell*)allocate_clear(sizeof(Cell));
    main_cell->name = copy_string("Main", NULL);
    lib.cell_array.append(main_cell);

    Polygon* rect = (Polygon*)allocate(sizeof(Polygon));
    *rect = rectangle(Vec2{0, -10}, Vec2{150, 10}, 0);
    main_cell->polygon_array.append(rect);

    Reference* ref1 = (Reference*)allocate_clear(sizeof(Reference));
    ref1->init(grat1);
    ref1->origin = Vec2{length, 0};
    ref1->rotation = M_PI / 2;
    main_cell->reference_array.append(ref1);

    Reference* ref2 = (Reference*)allocate_clear(sizeof(Reference));
    ref2->type = ReferenceType::Cell, ref2->cell = grat2, ref2->origin = Vec2{150 - length, 0},
    ref2->rotation = -M_PI / 2, ref2->magnification = 1, main_cell->reference_array.append(ref2);

    lib.write_gds("pcell.gds", 0, nullptr);
    lib.free_all();

    {
        ErrorCode error_code = ErrorCode::NoError;
        Library lib_three = gdstk::read_gds("pcell.gds", 0, 1e-1, nullptr, &error_code);
        if(error_code != ErrorCode::NoError)
            std::cout << "Error reading..." << std::endl;
        
        Array<Cell*> top_cells = {};
        Array<RawCell*> top_rawcells = {};
        lib_three.top_level(top_cells, top_rawcells);
        
        Array<Reference*> removed_references = {};
        for (uint64_t i(0); i<top_cells.count; ++i){
            Cell* this_cell = top_cells[i];
            this_cell->flatten(true, removed_references);
        }
        top_cells.clear();
        top_rawcells.clear();
        removed_references.clear();
        lib_three.free_all();
    }

    return 0;
}
```

I went to check the cell.cpp. If instead of 
```
            reference_array.remove_unordered(i);
            result.append(ref);
            ref->get_polygons(apply_repetitions, false, -1, false, 0, polygon_array);
            ref->get_flexpaths(apply_repetitions, -1, false, 0, flexpath_array);
            ref->get_robustpaths(apply_repetitions, -1, false, 0, robustpath_array);
            ref->get_labels(apply_repetitions, -1, false, 0, label_array);
```
I do:
```
            reference_array.remove_unordered(i);
            ref->get_polygons(apply_repetitions, false, -1, false, 0, polygon_array);
            ref->get_flexpaths(apply_repetitions, -1, false, 0, flexpath_array);
            ref->get_robustpaths(apply_repetitions, -1, false, 0, robustpath_array);
            ref->get_labels(apply_repetitions, -1, false, 0, label_array);
            free_allocation(ref);
```

The leak is fixed.

Am I using the flatten correctly or there is really a leak?

Here is the leak:
[image]

heitzmann, 2024-09-06:
Hi @Roy-974 sorry for taking so long to respond. I must have lost the email notification here.
The idea here is that you are responsible for clearing the references returned by `Cell::flatten` (we don't want to assume how they were allocated in the first place, so it's the user's responsibility to free them). You have to clear each reference and deallocate them after flattening.

Roy-974, 2024-09-08:
Hello @heitzmann, I understand that the user needs to clear and deallocate the array of references.
Is `removed_references.clear()` not enough to clear and deallocate the references?

heitzmann, 2024-09-09:
You still need to free the references themselves: free_alocation(ref) for
each ref in removed references.

Roy-974, 2024-09-10:
Ah, I see. I confirm that it works.
For some reason, I thought that when you cleared the array, it was deallocating the pointers inside the array.
Thanks!