heitzmann/gdstk #283: Some legacy GDS properties read in from OASIS file have either "!" or "1" appended to them

fnJeff, 2024-12-05:
Unfortunately, I have another weird one.  For some reason after reading in an OASIS file, I'm seeing that for a few reference/path/polygon's their respective GDS properties seem to have either a "!" or "1" appended to them.  I can not figure out what is different about these that would cause this.  I don't believe it has to do with the name in the property because there are other very similar names that are imported correctly.  I am 99% sure this isn't already in the OASIS for a couple reasons.  One, my layout tool can read back in the OASIS file and also write it out as a text format, and I don't see these "!" or "1" appended anywhere in this file.  Two, I can also use calibredrv to read in the OASIS file and can view the properties there and they also do not have these characters appended.

I put together as simple a testcase as I could to show that this seems to be happening either during reading the OASIS in or in some step triggered by that.  This contains both the OASIS and the human readable version of it, both generated from my layout tool.

[test.oasis.tar.gz](https://github.com/user-attachments/files/18026426/test.oasis.tar.gz)

I wrote the following script to simply identify all the references, paths, and polygons that have a GDS property 42 with names that have had this strange "!" or "1" appended to them upon being read in.  As you can see, the only thing being done after reading in the OASIS is accessing the names inside of the GDS property 42 using the get_gsd_property() method.  So that has to narrow down the hunt.
```
import re
import gdstk

lib  = gdstk.read_oas("test.oasis")
cell = lib.cells[0]

pattern = re.compile("^.+\\d_[1!]$")

sf = round(lib.unit / lib.precision)

out = list()

for prop_type in ("reference", "path", "polygon"):
    for obj in [obj for obj in getattr(cell, prop_type + 's') if obj.get_gds_property(42) != None and pattern.match(obj.get_gds_property(42))]:
        out.append("")
        ii = len(out) - 1

        match prop_type:
            case "reference":
                x = obj.origin[0]
                y = obj.origin[1]
            case "path":
                x = obj.spine()[0][0]
                y = obj.spine()[0][1]

                layer    = obj.layers[0]
                datatype = obj.datatypes[0]
            case "polygon":
                x = obj.points[0][0]
                y = obj.points[0][1]
                
                layer    = obj.layer
                datatype = obj.datatype

        x_int = round(x * sf)
        y_int = round(y * sf)

        out[ii] += f"{prop_type:>9} : {obj.get_gds_property(42)}"
        out[ii] += f"  --  approx location : ({x:7.4f}, {y:7.4f}) / ({x_int:6}, {y_int:6})"

        match prop_type:
            case "path" | "polygon":
                out[ii] += f"  --  layer : ({layer:>4}:{datatype:>4})"

for line in out: print(line)
```

The important output you should see after running this is as follows.  Note that I output the x/y location first in lib.units and then scaled into integer coordinates.  The integers are easily found in the OASIS text file for cross referencing.  What you will see is that none of these GDS properties contain the appended "!" or "1" shown below in the OASIS text file.  Hopefully you can easily determine what is causing this.
```
reference : calana_avg_offset_reg_4_!  --  approx location : ( 6.1560,  1.0800) / ( 12312,   2160)
reference : calana_avg_offset_reg_3_!  --  approx location : ( 6.1560,  0.6000) / ( 12312,   1200)
reference : calana_avg_offset_reg_2_!  --  approx location : ( 6.9540,  3.4800) / ( 13908,   6960)
reference : calana_avg_offset_reg_1_!  --  approx location : ( 6.6120,  3.4800) / ( 13224,   6960)
reference : calana_avg_offset_reg_0_1  --  approx location : ( 2.6220,  1.0800) / (  5244,   2160)
reference : FC_cg_cycle_count_reg_1_!  --  approx location : ( 4.4460,  4.4400) / (  8892,   8880)
     path : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.3680,  1.9600) / ( 10736,   3920)  --  layer : (  34: 330)
     path : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.0950,  1.9600) / ( 10190,   3920)  --  layer : (  34: 330)
     path : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.2360,  4.5200) / ( 10472,   9040)  --  layer : (  34: 330)
     path : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.2125,  4.5200) / ( 10425,   9040)  --  layer : (  34: 330)
     path : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.3190,  3.8800) / ( 10638,   7760)  --  layer : (  34: 330)
     path : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.3680,  3.8800) / ( 10736,   7760)  --  layer : (  34: 330)
     path : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.3200,  3.0000) / ( 10640,   6000)  --  layer : (  34: 330)
     path : clk_FC_cg_avg_dac_reg_0_1  --  approx location : ( 7.3920,  2.9300) / ( 14784,   5860)  --  layer : (  34: 330)
     path : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.3200,  1.9600) / ( 10640,   3920)  --  layer : (  35: 360)
     path : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.3200,  3.8800) / ( 10640,   7760)  --  layer : (  35: 360)
     path : clk_FC_cg_avg_dac_reg_0_1  --  approx location : ( 5.3200,  3.0000) / ( 10640,   6000)  --  layer : (  35: 360)
  polygon : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.3080,  1.9400) / ( 10616,   3880)  --  layer : (  34: 330)
  polygon : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.2800,  1.9200) / ( 10560,   3840)  --  layer : (  34: 330)
  polygon : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.3010,  1.9410) / ( 10602,   3882)  --  layer : (  54: 350)
  polygon : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.2820,  1.9220) / ( 10564,   3844)  --  layer : (  35: 360)
  polygon : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.1910,  4.4900) / ( 10382,   8980)  --  layer : (  34: 330)
  polygon : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.2800,  4.4800) / ( 10560,   8960)  --  layer : (  34: 330)
  polygon : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.3010,  4.5010) / ( 10602,   9002)  --  layer : (  54: 350)
  polygon : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.2820,  4.4820) / ( 10564,   8964)  --  layer : (  35: 360)
  polygon : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.2800,  3.8400) / ( 10560,   7680)  --  layer : (  34: 330)
  polygon : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.3010,  3.8610) / ( 10602,   7722)  --  layer : (  54: 350)
  polygon : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.2820,  3.8420) / ( 10564,   7684)  --  layer : (  35: 360)
  polygon : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.3380,  3.8400) / ( 10676,   7680)  --  layer : (  34: 330)
  polygon : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.2800,  2.9600) / ( 10560,   5920)  --  layer : (  34: 330)
  polygon : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.3010,  2.9810) / ( 10602,   5962)  --  layer : (  54: 350)
  polygon : clk_FC_cg_avg_dac_reg_0_!  --  approx location : ( 5.2820,  2.9620) / ( 10564,   5924)  --  layer : (  35: 360)
  polygon : clk_FC_cg_avg_dac_reg_0_1  --  approx location : ( 7.3320,  2.9600) / ( 14664,   5920)  --  layer : (  34: 330)
```

fnJeff, 2025-02-07:
This issue is really starting to cause me problems.  My workaround for trying to detect these conditions is colliding with actual names and I'm ending up with invalid names.

Has anyone been able to take a look at the test case I uploaded to see what is causing these strange characters to be suffixed on the names?

heitzmann, 2025-02-07:
Hi Jeff, I've just pushed a possible fix. Running your example code, I don't see any property errors. Can you check too?

fnJeff, 2025-02-10:
Hi Lucas, thanks for the possible fix!  I have to apologize for being a moron, because I have no idea how to test your fix.  Up until this point I have only used pip to install your official released versions, with 0.9.58 being the one I currently have.  I tried to understand the build from source instructions on the readme, but I have no idea what wheels are, or exactly how to get a version with your fix in it to install and check.

Can you offer a stupid hardware guy any help here?  Or I have to figure out how to compile it from the source?  I know I can not indiscriminately install things on my work unix machines, so I'm not sure if I can or can't do this locally.  Sorry, and thanks in advance for your help.

fnJeff, 2025-02-10:
I have no idea what went wrong, but I did at least try to download main code and build whatever a wheel is.  This is usually what happens any time I try to start from source code on anything.

[image]

heitzmann, 2025-02-10:
That shows you're in the right direction. You're missing development version of the qhull library in your system as a dependency. The other one that is required is zlib, but that is usually installed by default.

If you can't test, I can make a release anyway and see what happens...

fnJeff, 2025-02-11:
I don't think I will be able to make any changes to get this to compile.  Disadvantages to working in a trillion dollar company... no flexibility.  These are IT managed LSF hosts or simple virtual machines for running VNC, neither of which they will allow users to modify in any way.

Sorry for this annoying limitation on my end, but if you can make a release, I can test it right away.

heitzmann, 2025-02-11:
No problem. I've just triggered a release for you to check when it's done: https://github.com/heitzmann/gdstk/actions/runs/13274234846

fnJeff, 2025-02-12:
Hi Lucas.  I can confirm your fix for this issue in 0.9.59.

I'm not seeing the strange property name trailing characters anymore.  This isn't an exhaustive test, but I can definitely say it fixed the error I was getting in an OASIS file I had run my script on that was caused by this.  I was also able to remove my temporary workaround code that attempted to remove them, and from what I can see, everything looks good.

Thank you!  I really appreciate your time and effort on this project.

closed 2025-02-12 by heitzmann