heitzmann/gdstk #289: read_gds() doesn't work on latest release 0.9.59

abhishekchak52, 2025-02-14:
Hi, 

Firstly, thank you very much for your work on this wonderful package! 

I've run into an issue with the recent release where I cannot use `read_gds()` at all. It tells me that there is no attribute `read_gds` of `gdstk`. I'm not sure what's causing this, but the previous release (0.9.58) doesn't produce this error for me.

heitzmann, 2025-02-14:
What OS and python version are you on?
Also, can you share the full error message and of possible the installation log when you pip install hdstk?

abhishekchak52, 2025-02-19:
It was installed using conda. Somehow, downgrading to 0.9.58 and then back up to 0.9.59 has fixed the issue. I apologize for the confusion.

closed 2025-02-19 by abhishekchak52