heitzmann/gdstk #296: When rotating an imported layout some polygons go to the wrong place

bcerjan, 2025-04-02:
I have an inverse-designed grating coupler that I am reading in to the python interface to gdstk (from an .oas file). It works correctly:
[image]

I then want to rotate it to align with other structures that are drawn using gdstk. I noticed that when I rotate my structure (to any angle), I get some weird pixels in the wrong place:
[image]

For reference, the test code I'm running is:
```
...
lib = gdstk.read_oas('./coupler.oas')
shapes = []
for cell in lib.cells:
    for poly in cell.polygons:
        shapes.append(poly)

lib = gdstk.Library()

cell = lib.new_cell("TOP")

#shapes = [s.rotate(np.pi/2) for s in shapes] # uncomment for rotated
cell.add(s for s in shapes)

lib.write_oas("test.oas")
```

This matters for my use case as I'm calculating the overall bounding box of the coupler and those weird points throw off the calculation.

nmz787, 2025-04-21:
The documentation states the `GDSII` format's polygons with holes are not `directly` supported, so I would guess the functions in `gdstk` do not make further assumptions about whether a polygon has holes or not. See first few paragraphs here: https://heitzmann.github.io/gdstk/gettingstarted.html

Here's another approach though, to accomplish your goal, just `fillet` the rects before performing the `boolean` :
```
import gdstk
lib = gdstk.Library()
cell = lib.new_cell('test')

# polygons
rect = gdstk.rectangle((0, 0), (100, 100))
rect_2 = gdstk.rectangle((20, 20), (80, 80))
polys = gdstk.boolean(rect, rect_2, 'not', layer=1, datatype=1)
cell.add(*polys)

# polygons with round corners
r1_f = rect.fillet(20).translate(200, 0)
r2_f = rect_2.fillet(20).translate(200, 0)
fillet = gdstk.boolean(r1_f, r2_f, 'not', layer=2, datatype=1)
cell.add(*fillet)

lib.write_gds("test_polygonFillet.gds")
```

nmz787, 2026-02-21:
Can you supply the OAS?

bcerjan, 2026-02-21:
[coupler.zip](https://github.com/user-attachments/files/25460720/coupler.zip)

This is the coupler from above.

closed 2026-02-27 by bcerjan