heitzmann/gdstk #305: [Question] Is it possible to use gdstk to read huge OAS files (26GB) and write out cells in LEF/DEF format?

pallavg, 2025-04-29:
My problem statement is as follows:
1. I have very large OAS files (26GB) representing an IC design.
2. I would preferably like to selectively open certain cells (names are known apriori) instead of reading the entire file into memory
3. For each of these cells, I would like to iterate through their OASIS data and write it out in LEF/DEF format. I can write this converter myself.

I came across this library as a potential solution. I would like to explore it but I had a few questions first.

My questions:
1. What are the machine memory requirements would I need to read in such a design? Is 128GB enough or needs 256/512 GB? A user in #280 has posted his experience but was wondering if others had further insight?

2. Instead of reading the entire contents into memory, is it possible to read certain cells only and iterate through their contents?

Thanks.

heitzmann, 2025-04-29:
Unfortunately, no. The library reads the full file contents at once. However, you could take a look at the OASIS loading function and rewrite an iterative version. The difficulty with OASIS (that doesn't happen in GDSII) is that you'll have to read the whole file in sequence, even if you discard most cells, because of the state variables that get set along the way. Other than that, it should be somewhat easy to read only selected cells.

pallavg, 2025-04-29:
Thank you for the response. I will investigate the library further.